cmower/optas #38: Optimize subset of joint angles

joaomoura24, 2022-11-21:
At the moment i do not see an obvious way of optimizing declaring only a subset of the joint angles for optimization.

Imagine that you load a dual arm system, but you only want to optimize the motion of one of the arms.

The issue is that the OptimizationBuilder declares all the joints as optimization variables, which is not desirable in this specific case.

Any idea?

joaomoura24, 2022-11-21:
I have just had an idea that i will test tomorrow.
I'll close this if that works.

cmower, 2022-11-21:
Reopening, I had this thought before about the user being able to specify which are "planning" joints - maybe some joints need to be fixed or parameterizable (like the RJ mode in my CASE paper), or, as you say, may be part of a dual/multi arm system. 

Please share your approach when you've developed it and we can think about integrating.

joaomoura24, 2022-11-29:
I already added this functionality to my own repo in the [add-se_opt_joints](https://github.com/joaomoura24/optas/tree/add-sel_opt_joints) branch.
It works for the full IK example and am going to test for the QP case next, then I'll do the pull request.

There were some hard decisions in terms of the naming and where to put things, you might want to take a look to discuss before attempting to merge it.

cmower, 2023-02-25:
This has been addressed.

closed 2023-02-25 by cmower