cmower/optas #65: Add support for floating-base (6-DOF) joint type and prismatic joint (1-DOF).

duISIR, 2023-02-10:
I will use optas for the motion generation for mobile manipulator: EVA. However, optas currently only supports revolute and continuous joints. Could you add the support for the floating-base joint and prismatic joint? Thank you very much.

cmower, 2023-02-11:
Hi @duISIR, this is something that I have wanted to add but didn't have a need in my own work - nice to know you're using optas! I have started the implementation for handling prismatic joints (see #67), I will try in the next few weeks to finalize this and merge. I will also try to add support for floating-based joints too, but if you are able to contribute this feature please setup a pull request!

cmower, 2023-02-16:
I don't have the bandwidth to implement support for floating-base joints. The following methods are where you will need to implement the model. If you're able to implement, please also add an example with a floating-base robot to `example/` with your pull request.

The `get_global_link_transform` method.
https://github.com/cmower/optas/blob/937db570af15999439073e007bbffef558d2178b/optas/models.py#L495-L530

The `get_global_link_quaternion` method.
https://github.com/cmower/optas/blob/937db570af15999439073e007bbffef558d2178b/optas/models.py#L596-L628

The `get_global_link_geometric_jacobian` method
https://github.com/cmower/optas/blob/937db570af15999439073e007bbffef558d2178b/optas/models.py#L676-L734

cmower, 2023-04-26:
I am closing this issue for now. Please feel free to reopen if support for floating-base joints is required.

closed 2023-04-26 by cmower