cmower/optas #66: Issue with differential IK

cmower, 2023-02-10:
@joaomoura24, I am a bit confused with an example I'm trying to setup re differential IK. Could you take a look at the file: https://github.com/cmower/optas/blob/dev-diff-ik-example/example/linear_differential_ik_controller_example.py

When you run it using the position control, you will see that the robot doesn't keep up with the goal sphere. I expect some error but not as much as you can see in the example. When you change to velocity control, it does something but the motion doesn't look quite right for such a simple problem.

I wanted to setup a differential IK example using a QP solver on the robot that solves

```
  dq* = argmin || J(qc)*dq - vg||^2  s.t.  q- <= qc + dt*dq <= q+
```

where `J` in this case returns the 3-by-ndof linear part of the Jacobian and `vg` is the goal linear velocity.

Do you have any thoughts?

joaomoura24, 2023-02-12:
Apart from some undetected bug, I don't see anything fundamentally wrong. (I didn't analyse it very throughly)

The only concern is how are you computing the end-effector velocity.
If that velocity is unfeasible and you are adding it as an equality constraint, that can be trouble.

joaomoura24, 2023-02-12:
In my planar idk example I also put the forward differential kinematics and a equality constraint, but since then I changed it for a cost to avoid situations where the requested task space velocity is unrealistic.

cmower, 2023-02-25:
Thanks @joaomoura24, I think this is probably not an issue that needs fixing and just a case of problem specification.

closed 2023-02-25 by cmower