cmower/optas #77: TypeError: __init__() got an unexpected keyword argument 'layout'

joaomoura24, 2023-02-17:
When running the point_mass_planner.py from the master branch i get the following error to do with the Animate() method:

`Traceback (most recent call last):
  File "example/point_mass_planner.py", line 168, in <module>
    main()
  File "example/point_mass_planner.py", line 165, in main
    Animate(animate).show()
  File "example/point_mass_planner.py", line 99, in __init__
    self.fig, self.ax = plt.subplot_mosaic([['birdseye', 'position'],
  File "/home/joao/.local/lib/python3.8/site-packages/matplotlib/pyplot.py", line 1519, in subplot_mosaic
    fig = figure(**fig_kw)
  File "/home/joao/.local/lib/python3.8/site-packages/matplotlib/pyplot.py", line 797, in figure
    manager = new_figure_manager(
  File "/home/joao/.local/lib/python3.8/site-packages/matplotlib/pyplot.py", line 316, in new_figure_manager
    return _backend_mod.new_figure_manager(*args, **kwargs)
  File "/home/joao/.local/lib/python3.8/site-packages/matplotlib/backend_bases.py", line 3544, in new_figure_manager
    fig = fig_cls(*args, **kwargs)
TypeError: __init__() got an unexpected keyword argument 'layout'
`

cmower, 2023-02-17:
This is because you have an old version of matplotlib installed. `pip install --upgrade matplotlib` should resolve this.

joaomoura24, 2023-02-17:
Great. That was it!

closed 2023-02-17 by joaomoura24