cmower/optas #84: Suggestion: deprecate optimize_time

cmower, 2023-02-21:
The goal of the `optimize_time` flag in the `OptimizationBuilder` interface was originally meant as a helper for when the user wanted to make `dt` part of the decision variables. In reality, this doesn't happen very much and in the case that it does, it is not such a big ask for the user. Furthermore, because of `optimize_time`, it convolutes the methods below, making it very difficult to follow the logic.

https://github.com/cmower/optas/blob/3eeeaf3309f75d6ffbc3b158055f651005b0e1d6/optas/builder.py#L563-L623

@joaomoura24, unless you have any major objections, I will deprecate this feature as part of the #37. I am guessing you probably aren't using this flag in your code?

joaomoura24, 2023-02-22:
I agree.
Yes, I am not using it in any of my examples, and there are many other ways of optimising time, so I think that specifying one is won't be that useful in general.

cmower, 2023-02-22:
Closing, this will be fixed when #37 is merged.

commit a301591a (2023-02-22): Deprecate optimize_time, fix #84
commit 39788876 (2023-02-22): Deprecate optimize_time, fix #84
closed 2023-02-22 by cmower