cmower/optas #111: SyntaxError: invalid syntax during Run tests after the installation

zhuhongwu0918, 2023-03-14:
I install OpTaS, and couldn't pass the test. I change the python version of 3.6.9 or 3.8. But this error is still exits.
../../.local/lib/python3.6/site-packages/_pytest/assertion/rewrite.py:170: in exec_module
    exec(co, module.__dict__)
tests/test_builder.py:2: in <module>
    import optas
optas/__init__.py:4: in <module>
    from .models import RobotModel, TaskModel
E     File "<fstring>", line 1
E       (time_deriv=)
E                  ^
E   SyntaxError: invalid syntax

cmower, 2023-03-14:
Hi @zhuhongwu0918 , thanks for raising this issue. I'm unable to reproduce the error. Could you let me know exactly what you run that produces this error, please also let me know the version of python and pip, and your OS.

cmower, 2023-03-14:
I have just looked into this further and I am quite sure that you are running an older version of Python that is not supported by OpTaS (i.e. Python 3.6). The error you show is coming from the use of f-strings in `builder.py`.

See this example

```
$ python3.6
Python 3.6.15 (default, Apr 25 2022, 01:55:53) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> a = 234
>>> print(f"Hello {a=}")
  File "<fstring>", line 1
    (a=)
      ^
SyntaxError: invalid syntax

$ python
Python 3.8.10 (default, Nov 14 2022, 12:59:47) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> a = 234
>>> print(f"Hello {a=}")
Hello a=234
```

Probably the best thing is to upgrade your version of Python to 3.8 and re-run the tests - make sure `python` is linked to the correct version of Python.

cmower, 2023-03-14:
Please let me know if this resolves your issue?

cmower, 2023-04-02:
I am closing this issue. 

I am quite certain that the issue you are experiencing is due to the version of Python you are running (i.e. 3.6). Since the latest merge #119, we are testing support for 3.7>=. 

Please consider upgrading your version of python. 

If the issue persists with a later version of python (i.e. 3.7, 3.8, or 3.9) then please re-open.

closed 2023-04-02 by cmower