cmower/optas #124: Issue with LBR model.

cmower, 2023-04-10:
I'm not sure when this happened, but the LBR model in the examples is messed up. From running other code, I am pretty much certain there is nothing wrong with `optas.RobotModel`. I think the issue must be in `optas.Visualizer`. Strange that the LWR is fine. It could be something to do with how this visual frames are defined (I made changes in the visualizer that handled the case when multiple visual frames for a link are defined). When I load the same model for the LBR into Pybullet it is put together fine. I probably missed something standard.

FrankC96, 2023-06-11:
You should consider removing all the "arrayfying" decorators and put a clause like so:

```
if isinstance(position, DM):
            position = position.toarray().flatten().tolist()
        else:
            position = position
```

for all variables that are converted to a list in visualize.py. This is a quick fix to run the example, I don't know how it will affect the whole program.

cmower, 2023-06-16:
Thanks @FrankC96! The `arrayify` decorator is to ensure that the array-like inputs to all the decorated functions have type `casadi.DM` or `casadi.SX` - the main point of this is to prevent code repetition.

Just to clarify, did you have the changes to `visualize.py` that make the LBR example work without affecting the other examples? If so, I'd appreciate if you could set up a pull request.

FrankC96, 2023-06-16:
To my best knowledge the other examples, are not affected. I created the pull request.

cmower, 2023-08-11:
[image]

I loaded the LBR model into pybullet, and all is well. This confirms the issue is with the visualizer and how it extracts the robot geometry.

RLi43, 2025-01-22:
Hello, 
I think the issue is there's something wrong when the robot model is built from URDF. I tried some joint positions and the FK results were not reasonable. I suspect that the joint's axis is not considered in transformation.