cmower/optas #143: Installation - sudo pip install .

sandorfelber, 2023-06-08:
Tried pulling the latest master branch using git pull, and then pip install . from the directory, but then got "AssertionError: given link RARM_tool_link doesn't appear in URDF" on Linux Ubuntu Focal 20.04 machine and ROS Noetic.
See traceback message [here: https://drive.google.com/file/d/1RBSWqI1PU8tPPe_h293daTd7nKjsozF3/view?usp=sharing](https://drive.google.com/file/d/1RBSWqI1PU8tPPe_h293daTd7nKjsozF3/view?usp=sharing).
_sudo_ pip install . downloaded and installed among other dependencies, the URDF parser dependency, which has resolved the issue.

cmower, 2023-06-08:
Thanks @sandorfelber. Sorry, I'm confused, is this issue is resolved for you?

The error suggests that the link `RARM_tool_link` doesn't exist in the URDF, so probably a good idea to double check your URDF/xacro file.

Could you please elaborate if you are still seeing errors, or can we close?

Also, it's **not** a good idea to use `sudo pip install`, see [here](https://askubuntu.com/a/802594).

sandorfelber, 2023-06-08:
Yes, I have resolved the issue by sudo installing optas. I talked to Joao and he suggested I raise an issue here regardless.

I spent an hour or so yesterday verifying the URDFs in an effort to debug the error posted, but the URDFs and xacros all looked correct (these files are the same as Joao`s working implementation that ran several times on this same system), which then led me to think it`s an optas parsing issue.

Errors are gone and it`s working - I understand sudo pip isn`t the best approach by default, but I thought it might be worthwhile mentioning that the library failed at pulling/downloading and installing the necessary dependencies on that machine without the elevated permissions.

cmower, 2023-06-08:
Ok, thanks. 

I'm a little confused why this would happen, and even more confused why `sudo pip install` resolved the issue. I wonder if parts of the URDF were not in the path visible to the URDF parser and that's why it complains, do you have multiple xacro files that make up the robot? Perhaps by elevating the privileges the required paths are added. I tried re-creating the issue but everything worked as expected - I just tested some of the examples. So I am inclined to think the issue is with the URDF you are using - see [known issues](https://github.com/cmower/optas#known-issues), perhaps what you saw is somehow related to the first point also.

Thanks for raising the issue, it is useful to have these details noted. 

I will close for now since you have resolved the issue, but please re-open if it happens again or if you have any more ideas why this happened and potentially a more suitable fix if others experience the same issue.