cmower/optas #157: Possible bug in models.py

OmidRezayof, 2024-06-26:
Hi,
Great job with this amazing library!
I wanted to inform you about a possible bug in [models.py](https://github.com/cmower/optas/blob/master/optas/models.py)

```
J = self.get_global_link_geometric_jacobian(link, q)

        # Transform jacobian to given base link
        R = self.get_global_link_rotation(base_link, q).T
        O = cs.DM.zeros(3, 3)
        K = cs.vertcat(
            cs.horzcat(R, O),
            cs.horzcat(O, R),
        )
        J = K @ J
```

In line 1340 and when trying to transfer a Jacobian matrix from the world link to a desired link, matrix 'K' (also known as the Adjoint of the Transformation between the world and the desired link) should look something like this:

[image]

which made me think there might be a bug in the code. If you're using a different logic, please accept my apology and let me know.
Thanks.

cmower, 2024-06-26:
Hi @OmidRezayof, many thanks for your interest in optas and raising this issue. It is possible there is a bug however I don't remember noticing any issues with this functionality in the past. Have you noticed any incorrect results in any controllers/planners using optas?

OmidRezayof, 2024-06-27:
I have not. I was looking at your library and noticed this typo :)
I'm not sure if we're using the same logic behind this transform but I would encourage you to take another look into it since with my knowledge that might have problems when you actually use it to transform Jacobians from space to another frame.
Just wanted to raise this to your attention.